bigtreetech/OctoBTT #2: In install.sh wrong dir.

Tompudex, 2020-04-25:
In a file;
sudo service octoprint stop
cp OctoBQPi/config.yaml .octoprint/config.yaml
sudo service octoprint start
cp OctoBQPi/.xsession .xsession
the path is incorrect.

Minglarn, 2020-04-26:
I've tried to change to right path, but it complains about OctoBTT/.xsession is not found. 
Any solution for that?

bigtreetech, 2020-04-27:
> In a file;
> sudo service octoprint stop
> cp OctoBQPi/config.yaml .octoprint/config.yaml
> sudo service octoprint start
> cp OctoBQPi/.xsession .xsession
> the path is incorrect.

Thank you for your feedback. I have found the problem. I am sorry for the omission of project file upload.

bigtreetech, 2020-04-27:
> I've tried to change to right path, but it complains about OctoBTT/.xsession is not found.
> Any solution for that?

This problem is the omission of the project file upload on my side. At present, we have added the missing file. Thank you for your feedback.

closed 2020-04-27 by bigtreetech