bigtreetech/OctoBTT #4: Camera icon does not work

Poppy45, 2020-08-04:
When pressing the camera icon opens a blank page .can see camera output on web page of Octoprint. Any help

skinz98, 2020-08-06:
Having the same problem

Shine6Z, 2020-08-11:
Thank you for your exploration. As I am still testing this section, it will not be online for the time being, and these functions will be put online one after another after it becomes stable.

byseven07, 2020-08-21:
Same issue !!!

3dbret, 2020-09-19:
this does not work for me too, it does nothing.

bigtreetech, 2020-09-19:
Until now, the rendering code of video stream cannot work properly, so this meaningless code fragment cannot be opened to users for the time being. In the latest update, I will close this entry until the problem is completely solved and it will be opened to users again. I am very sorry for the misunderstanding caused by everyone.

bigtreetech, 2020-10-23:
Thanks for waiting, now Octoprint's previews are supported and you're welcome to try out the new feature.

closed 2020-10-23 by bigtreetech