bigtreetech/OctoBTT #6: What does the "Ban" icon refer to

anupchavda1, 2020-08-07:
I'm a newb when it comes to programming/coding, but i had a look at the c++ code and noticed that the word "BAN" related to something to do with the Filament...But the word does not make sense in this context. looking further it looks like its the connection / printer status.

can we get a more appropriate label for the status rather than the word "BAN"?

Thanks

Shine6Z, 2020-08-11:
Ok, I will choose a more reasonable display to display the place, for now I just think this icon is used to remind the user that it cannot be operated or not recommended.

bigtreetech, 2020-10-23:
A new version has been released and I have changed the Busy" Ban" display status to "Busy". Thanks for your feedback.

closed 2020-10-23 by bigtreetech