bigtreetech/OctoBTT #8: Wifi not working with the Screen Connected

fenix8k, 2020-08-20:
I just get the new BIGTREETECH PI TFT50 V1.0 TFT-display, it work really well, buuuut, when I have it attached to a Raspberry Pi 4 with the mounting screws, the Wifi drops or work really bad, if I detach the Screen from the Pi (still attached to the pi via the cable), the Wifi start working Again, any sugestión?, I read that HDMI signal can interfiere with 2G and 5G networks this can explain this behavior.

Thx!

bigtreetech, 2020-08-28:
The question is under confirmation, please wait patiently, and we will reply in two weeks.

Shine6Z, 2020-08-31:
We passed the preliminary check problem for Raspberry Pi 4 Model B Wifi problems, Raspberry Pi 4 Model B antenna radiation direction to the main board PCB back, when the PCB back a valid GND can form shielding effect of antenna, leading to insufficient antenna performance and the disturbance, the current solution is to use 17.5 mm high column screw to hold down the blackberry pie motherboard and the spacing between the screen, so that antenna radiation reserve enough space, can work stable.

closed 2020-09-03 by bigtreetech

NovaViper, 2021-03-08:
I'm having the same issue, will this be fixed????

Acidice1000, 2022-01-08:
Pi4 with pi tft 50 v2. The moment the ribbon cable is plugged in the pi does not connect with wifi. Used a 2m cable and still it is not working. Tried with pi 3 and same issue. Problem is definitely not fixed.

hoonu, 2022-01-26:
Thanks for the heads up. I was trying to decide between the Pi 7" and this screen and it seems this might just be the deciding factor. I wonder if they just have some lcds out there that spew more EMI than others and that's why this issue isn't as widespread.

Hemmertyme, 2022-02-11:
I just returned my TFT35 v3.0 because it wouldn’t work with Klipper.   I didn’t return my SKR mini E3 because I had hopes for another option.  So I decided to order the TFT50 as a screen for my pi.  It arrived and now this damn thing doesn’t work either.  I have wasted sooo much time dealing with BTT’s shoddy products.  Unless I get some resolve I’m done with their products.