bigtreetech/OctoBTT #9: Please remove OctoDash assets

UnchartedBull, 2020-08-24:
Please remove all OctoDash assets from this repository (like https://github.com/bigtreetech/OctoBTT/blob/master/assets/icon/icon-alt.svg). Since you haven't kept the original LICENSE you are not allowed to use them within your "inspired by OctoDash" program. 

I would also advise you to remove the other icons you copied as well, since you also need to mention the auther / origin of them as indicated by their license from thenounproject.

@bigtreetech You guys could've just contacted me and we could've worked something out so you could use OctoDash with custom branding and stuff - I'm still happy to talk to you guys about that, so just let me know. But please do remove the copied assets, which you don't have a license for.

Shine6Z, 2020-08-28:
Well, we will immediately remove OctoDash from OctoBTT's database. I am very sorry for the effect this has had on you and we will also remove the relevant non-authorized resource, thank you for your reminding.

bigtreetech, 2020-08-28:
The associated resource has been removed.

closed 2020-08-29 by UnchartedBull