bigtreetech/OctoBTT #12: Screen Sleep

darouss99, 2020-09-04:
Hi, 

Is there a way to configure the screen to turn off itself after 30 minutes of inactivity?  If not, it will be a great feature to have inside OctoBTT, it will save the life of the screen.  I don't think a screen open 24/7 will have a long life.

Thank you

bigtreetech, 2020-09-08:
You can do that , setting profiles in file "**_/home/pi/.xsession_**" , 
change the second line , for example : "**xset -dpms**" >>> "**xset dpms 0 0 10**"  <<< Close the screen after 10 seconds

Parameter manual:
       xset dpms **_standby_** **_suspend_** **_off_**
       xset dpms force[**_standby_**|**_suspend_**|**_on_**|**_off_**]
       -dpms   The -dpms option disables DPMS (Energy Star) features.
       +dpms   The +dpms option enables DPMS (Energy Star) features.
       The  dpms option allows the DPMS (Energy Star) parameters to be set.  The option can take up to three numerical values, or the “force”flag followed by a DPMS state.  The “force” flags forces the server to immediately switch to the DPMS state specified.  The DPMS state can be one of ‘standby’ or ‘suspend’ , ‘off’, ‘on’  When  numerical values are given, they set the inactivity period (in units of seconds) before the three modes are activated.  The first value given is for the “standby”mode, the second is for the “suspend”mode, and the third is for the “off”mode.  Setting these values implicitly enables the DPMS features. A value of zero disables a particular mode.

darouss99, 2020-09-08:
Thank you for your answer.  I add "xset dpms 0 0 10" in my .xsession file to test.  It's working, the screen goes blanc but the backlight remain on.

Is there a way to turn off the backlight too?

Thank you

bigtreetech, 2020-09-08:
Sorry, backlight cannot be turned off completely because the screen backlight is not programmed to be turned off unless there is a separate backlight switch on the hardware to do so manually.

CBDesignS, 2020-09-08:
how do we mod the board to add a switch then ?? cutting a track and soldering a switch in sounds like a plan..

darouss99, 2020-09-08:
Ok, so I guess we can't dim the backlight too?  Maybe a great improvement for a next revision.  Is the programmation can be changed or if we need to modify the backlight to add a switch or a relay?

bigtreetech, 2020-09-10:
Indeed, the backlight brightness and backlight switch cannot be controlled by the program at present.You can remove the R6 0 Ω resistor, replaced by a sliding rheostat brightness control can be realized, if need to increase the switch, with a switch in the circuit in series can implement switch screen backlight.

darouss99, 2020-09-10:
I just tried it and it's work great. :)  I will wire it to a relay and when I will turn on my printer it will turn on the backlight at the same time.  Thank you for your help.

tobyfu, 2020-10-14:
it would be a useful feature to be able to turn the backlight on and off by software, I use klipper and the pi is on 24/7 which means the screen is also on and while I can make it go blank it is still very bright especially at night

bigtreetech, 2020-10-23:
> it would be a useful feature to be able to turn the backlight on and off by software, I use klipper and the pi is on 24/7 which means the screen is also on and while I can make it go blank it is still very bright especially at night

Yes, I will add your Suggestions in the future version, please stay tuned.

Shine6Z, 2020-10-23:
> it would be a useful feature to be able to turn the backlight on and off by software, I use klipper and the pi is on 24/7 which means the screen is also on and while I can make it go blank it is still very bright especially at night

I tried to connect the control pin of the backlight of the screen with PWM controlled IO to realize the backlight brightness control. This backlight adjustment function can be realized in the future.

SHexplorer, 2020-12-21:
> I tried to connect the control pin of the backlight of the screen with PWM controlled IO to realize the backlight brightness control. This backlight adjustment function can be realized in the future.
How did you manage to do that? Do you use a transistor to switch the two pins of R6 or do you just use 3V PWM?

ssavickiy, 2021-01-09:
there are not enough settings for the display brightness (backlight) and the option to completely turn off the backlight. Here is an implementation,
https://github.com/linusg/rpi-backlight

I have BIGTREETECH PITFT50 V1.0 installed and in idle moments, the display turns black, but the backlight still lights up.

Shine6Z, 2021-05-07:
> How did you manage to do that? Do you use a transistor to switch the two pins of R6 or do you just use 3V PWM?

just use 3V PW

Syco54645, 2021-07-22:
> I tried to connect the control pin of the backlight of the screen with PWM controlled IO to realize the backlight brightness control. This backlight adjustment function can be realized in the future.

Sorry I am not understanding here. Did you manage to get the backlight of the 1.0 screen to be controllable via Linux? If so can you please elaborate a bit? Where is the control pin for the backlight and what mechanism did you use software wise to actually control it?

Shine6Z, 2021-07-23:
> Sorry I am not understanding here. Did you manage to get the backlight of the 1.0 screen to be controllable via Linux? If so can you please elaborate a bit? Where is the control pin for the [...]

The IO pin of raspberry PI controls the screen brightness control point mentioned above (where the R6 resistor is located). Raspberry PI needs to control the IO pin in PWM mode through the program to realize the brightness control of the screen. The current program has not added IO control code, so the people in need need to implement this function by themselves.

eduncan911, 2021-12-15:
> @Shine6Z : The IO pin of raspberry PI controls the screen brightness control point mentioned above (where the R6 resistor is located). Raspberry PI needs to control the IO pin in PWM mode through the program to realize the brightness control of the screen. The current program has not added IO control code, so the people in need need to implement this function by themselves.

This is a good idea.  Without a diagram, I'll attempt to repeat what you said to see if I'm right.

> @bigtreetech : You can remove the R6 0 Ω resistor, replaced by a sliding rheostat brightness control can be realized, if need to increase the switch, with a switch in the circuit in series can implement switch screen backlight.

## remove the R6 0 Ω resistor

0Ω resistor basically means it's just a straight wire, interesting.  

So this, along with the other statement above, means Pulse Width Modulation (PWM).  which means we can operate it on a duty cycle of 0% (fully open) to 100% (fully closed), and any value in-between.

> @Shine6Z : just use 3V PW

I'm not following this part.  Are you saying set the RPi GPIO to PWM, at 3.3V?  So, the frequency doesn't matter?  Just as long as 50% duty just means 50% GND/VREF (need to measure)?  That sounds easy enough.

You also mention voltage, 3.3V.

So, can one guess that this may be a simple 1-wire hack to control backlight?  Take the PWM control, and connect it directly to the R6 location?  Is it VREF or GND?

Is the backlight powered by this voltage / do we need to worry about the power draw of the backlight from the PWM control circuit?

---

Full disclosure, I have a V2.0 display of the Pi TFT50 - which has a rheostat on it for brightness.  However, the 3d printed case I am using does not allow access to the knob.  So I wonder if it can be "bypassed" with this PWM control.