bigtreetech/OctoBTT #13: Error in OctoBTT

darouss99, 2020-09-04:
Hi, 

I just installed OctoBTT on a Raspberry Pi 3B+. OctoBTT open but I have "error" in the lower right of the screen.  I can not control the printer at all. The screen doesn't seem to detect OctoPrint.  Octoprint (1.4.2) run on the same Raspberry Pi and is working great with my CR-10s.

Is there any log I can look at to know what this error is about?

Thank you

bigtreetech, 2020-09-08:
Are you going to install Octopi mirror or manually compile OctoPrint?

darouss99, 2020-09-08:
Hi, Octoprint is already installed since about 2 years.  If I remember, I manually compiled Octoprint because the image (Octopi) wasn't compatible with the Pi 3b+ at this time.

Thank you

bigtreetech, 2020-09-08:
So, the problem might have been due to the path from which you installed it, and I'll try to follow up and try to fix it in a later version, but at the moment I'd only recommend using OctoPi to install to make sure it works, unless your OctoPrint compilation path and OctoPi are the same.