bigtreetech/OctoBTT #14: Autoconnect wrong baud rate

3dbret, 2020-09-19:
As it is this will not work with an Artillery Sidewinder X1 or any other printer with a 250000 baud rate.   This really needs to be a configuration option at install.

To fix:
change:
mainwindow.cpp:            _ConnectJson.insert("baudrate",115200); 
to this:
mainwindow.cpp:            _ConnectJson.insert("baudrate",250000);
do a make clean; make; qmake OctoBTT.pro;

bigtreetech, 2020-09-19:
In order to ensure the stability and anti-interference ability of the communication connection, it is recommended to use the intermediate 115200 Baud rate value. The default setting of the program is 115200 at present. More baud rate options and the communication mode of direct serial port will be obtained in future version updates.The make clean command can only work properly if it is run on the premise of existing compilation, otherwise the installation may be interrupted by unknown problems.Qmake octobtt.pro is required only if the list of items in the project file has changed, otherwise there is no sense in re-executing it.

3dbret, 2020-09-19:
ok thanks re. qmake but as stated it will not work, as in it will not
connect at all with printers that have a 250000 baud rate, like the
Artillery (Sidewinder X1) range.

bigtreetech, 2020-09-21:
Hello, you can modify the baud rate of the printer to 115200 so that OctoBTT can automatically connect to the printer.

3dbret, 2020-09-21:
that is not possible with the Artillery Sidewinder X1 and many
other printers, it is only 250000 therefore your software does not work
without modification and re-compilation.

ftcorey3, 2021-06-09:
Was this ever done, i have three printers that are affected by this bug. If you have not fixed it, can you post how a user can do this?