bigtreetech/OctoBTT #17: OctoBTT not starting after restart.

LittleDevil76, 2020-10-06:
Just installed OctoBTT on a new Rass Pi 4b now, followed your guide  and after ". install.sh" is finish, i have full control of the printer
on the 5" screen.
But, after i do a restart of the Pi, the OctoBTT do not start. It just loads up to command line, and thats it.
[image]

Is there something im doing wrong or what.??

bigtreetech, 2020-10-07:
Refer to Step 4 to check that the configuration is correct.

LittleDevil76, 2020-10-07:
Sorry, same deal. Does not start..hmm
But after i do a . install.sh , the install stops ¨freezes right here
[image]

it boots at this point and i can do whatever, but after a reboot, same deal..
Strange..!`?
[image]

bigtreetech, 2020-10-08:
> Sorry, same deal. Does not start..hmm
> But after i do a . install.sh , the install stops ¨freezes right here
> [image]
> 
> it boots at this point and i can do whatever, but after a reboot, same deal..
> Strange..!`?
> [image]

If it is true that starting the remote interface from a command does cause the command line to freeze, you can manually restart OctoBTT from the control page of the interface to break the freeze; Do not attempt to force an exit on the remote interface, as this will cause the interface to lock and refuse to start. If you refuse to start, you can only release the lock by restarting raspberry PI.

LittleDevil76, 2020-10-08:
This one`?
[image]

bigtreetech, 2020-10-09:
> This one`?
> [image]

Right, now I know what the problem is, your boot command is wrong, the correct command is `su -l pi -c "startx -- -nocursor"`
* **-l** (This is a "**Low**" lowercase '_L_’, not an '_I_' in "**Issue**")
* Reference images link ([https://github.com/bigtreetech/OctoBTT/blob/master/IMG/4_3_startup.png](https://github.com/bigtreetech/OctoBTT/blob/master/IMG/4_3_startup.png))

closed 2020-10-19 by bigtreetech