bigtreetech/OctoBTT #19: Icons and font have been downscaled?

frostydotjack, 2020-10-13:
Hi,

First, thank you for your work.

I previously had OctoBTT installed on a Raspberry Pi running octoprint until about a week ago, when I had to restore the sdcard to its former state. When it was running on the pi, it worked very well and performed great, no problems overall.
I have just got to reinstalling it tonight & when the ui appears, the font & all icons displayed are very small. I don't think it's the display I'm using. Any chance you know what's going on? More than happy to provide information to help. :)

I figured I should ask here, before doing anything else...

bigtreetech, 2020-10-15:
Thank you for your feedback:
This is certainly a problem that has a big impact on usage, but I haven't found it yet. As for your feedback, I will test and observe it in the near future. Once I find the problem, I will notify you here and I will fix it.

skinz98, 2020-10-25:
yea have same problem now, i first installed icons was a nice size now there a bit wee for me to see unless am really close

bigtreetech, 2020-10-26:
> yea have same problem now, i first installed icons was a nice size now there a bit wee for me to see unless am really close

Is it convenient for you to provide photos for me to have a look。

skinz98, 2020-10-26:
[Image][Image]

bigtreetech, 2020-10-31:
> [Image][Image]
> […](#)

Looks like your picture failed to upload.

frostydotjack, 2020-11-01:
[image]
[image]