bigtreetech/OctoBTT #21: Configs changed after install

Will-wastelander, 2020-11-07:
It appears that instead of keeping the users existing config, you guys back it up, and copy a new one in place of it. This can be bad for someone that has spent days getting their OctoPrint install setup and dialed in to how they want it. Instead, you guys should be adding the required lines to the config, if any that are needed to enable OctoBTT to work.

You can use this command to get additional info on how to set configs via CLI.

~/oprint/bin/octoprint config --help

And here is some info on what is in the config.yaml file.

https://docs.octoprint.org/en/master/configuration/config_yaml.html

Will-wastelander, 2020-11-07:
And for security reasons, you guys need to stop using the same API key for all installs. This is a HUGE security risk.

Will-wastelander, 2020-11-08:
Hardcoded API key... 
https://github.com/bigtreetech/OctoBTT/blob/e3516ea90a2b578b497829ca3616c09f1ad2241e/main.cpp#L17
https://github.com/bigtreetech/OctoBTT/blob/e3516ea90a2b578b497829ca3616c09f1ad2241e/octonetwork.h#L35

bigtreetech, 2020-11-10:
Yes, I am trying the way you said, but you still need to wait patiently before I debug and stabilize. Thank you for your feedback.

Will-wastelander, 2020-11-10:
There also seems to be an issue w/ Settings > WiFi, as when I click it, it either hangs OctoBTT, or drops to console.

Will-wastelander, 2020-11-10:
Also, OctoPrint will be making user auth mandatory in the near future, so it might be good to start getting that worked on.

bigtreetech, 2020-11-10:
The Wi-Fi connection is performed by the console operation mode, and I did not encounter the situation of falling into the console. As for the problem you encountered, I wonder if you could record a video for me, so That I can locate your problem more easily.
The second question is about the account permissions you think we will finish as soon as possible. Please wait patiently and thank you for your attention and support.

Will-wastelander, 2020-11-10:
@bigtreetech I can do the video, no problem. I have a spare rPi and screen I can set it up on. I should be able to get it in the next day or 2.

I am active user on OctoPrint's discord server, and have been told that auth will be mandatory in the near future. I believe it will be 0.15.0 that implements this requirement. 

I look forward to the official release. Will this be the software used w/ the Biqu BX rPi interface ? I am a kickstarter backer of it, and can't wait for the printer to be delivered. :)

bigtreetech, 2020-11-10:
Thank you for your support. I also have a request that you send me a copy of your configuration file, and I will refer to your configuration requirements to improve the usability of the software.

Will-wastelander, 2020-11-10:
What configuration file do you need ? And where can I locate it ?

Will-wastelander, 2020-11-10:
config.yaml for OctoPrint ?

Will-wastelander, 2020-11-10:
I have completed a fresh install of OctoPi 0.18.0 RC1 armhf. I completed a apt update and upgrade, then installed OctoDash before configuring OctoPrint. I then logged into OctoPrint, and did not get the normal Welcome wizard for the new installation. Here is the config.yaml that is currently in use. (The BTT Provided config.yaml I believe.) Along w/ the vanilla config that came with the install. If you need a live config, I will need to sanitize it before sending it to you. But all salts, sercret keys, api keys, application keys, discovery upnpUuid, and error tracking/tracking unique_id are generated uniquely per installation. They cannot be reused for another system, or this causes a security issue.

Also... https://octoprint.org/blog/2020/11/10/new-release-candidate-1.5.0rc1

[config.yaml.OctoBTTbackup.txt](https://github.com/bigtreetech/OctoBTT/files/5519715/config.yaml.OctoBTTbackup.txt)
[config.yaml.txt](https://github.com/bigtreetech/OctoBTT/files/5519716/config.yaml.txt)

Will-wastelander, 2020-11-10:
Here is the video for the wifi issues. It looks like it's just hanging, and doesn't do anything at all. I am running on a rPi 4 4GB.

https://drive.google.com/file/d/1QVfl7Mkzt5qdeotWIxBuVq4aAmaWnNR2/view?usp=sharing

Will-wastelander, 2020-11-10:
[image]

cp2004, 2020-11-10:
@bigtreetech - Just to let you know, as of [OctoPrint 1.5.0](https://github.com/OctoPrint/OctoPrint/releases/1.5.0rc1) OctoBTT no longer works, because of the configuration copying. It requires Access Control to be enabled, which also means you need a `users.yaml` which has encrypted passwords in it. As a result, users cannot open OctoPrint.

You **should not** need change the *entire* configuration like this to connect to OctoPrint. If you want an example of how it can be done, OctoDash is a good example. You **only need** the API key, not the entire file.

Will-wastelander, 2020-11-10:
There is no reason to sudo to read info either. sudo should **ONLY** be used for system changes, not reading data..

https://github.com/bigtreetech/OctoBTT/blob/e3516ea90a2b578b497829ca3616c09f1ad2241e/wlanconfig.cpp#L202

ifconfig | grep -E "flags|inet|ether"
eth0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether dc:a6:32:a4:xx:xx  txqueuelen 1000  (Ethernet)
lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.x.xxx  netmask 255.255.255.0  broadcast 192.168.x.255
        ether dc:a6:32:a4:xx:xx  txqueuelen 1000  (Ethernet)

Will-wastelander, 2020-11-10:
@bigtreetech @Shine6Z If I want to recompile my changes for testing, what do I need to run from the rPi ? The below lines ?

`cd ~/OctoBTT
qmake OctoBTT.pro
if [ ! -f "$OctoBTT" ]; then
 rm OctoBTT
 make clean
fi
make -j$(cat /proc/cpuinfo | grep processor | wc -l)`

Will-wastelander, 2020-11-11:
Hardcoding password, APIs, etc is a HUGE no no.. This needs to be removed.

https://github.com/Will-wastelander/OctoBTT/blob/bfffb6cbc50562a9bbd6eee861d62279f6417c44/terminaldialog.h#L54

Will-wastelander, 2020-11-11:
If needing sudo for the commands is required, which some of them aren't. You should be using something like this in /etc/sudoers.d/OctoBTT

pi ALL=(ALL) NOPASSWD: /sbin/iwlist, /sbin/wpa_cli, /sbin/iwconfig, /sbins/ifconfig

Will-wastelander, 2020-11-11:
I have forked the build, and will make some security changes, and do a PR. I'm not that into coding, so I won't be much help with anything outside of security stuff, and recommendations..

bigtreetech, 2020-11-11:
> I have completed a fresh install of OctoPi 0.18.0 RC1 armhf. I completed a apt update and upgrade, then installed OctoDash before configuring OctoPrint. I then logged into OctoPrint, and did not get [...]

Yes, I have noticed the existence of similar problems, so I will solve this problem as soon as possible.

bigtreetech, 2020-11-11:
For incorrect console passwords, you can enter the correct console password by clicking on the console icon at the top.

bigtreetech, 2020-11-11:
> @bigtreetech - Just to let you know, as of [OctoPrint 1.5.0](https://github.com/OctoPrint/OctoPrint/releases/1.5.0rc1) OctoBTT no longer works, because of the configuration copying. It requires [...]

Indeed, here I do need to adjust for the right access, I'm already doing it, you need to be patient.

bigtreetech, 2020-11-11:
> There is no reason to sudo to read info either. sudo should **ONLY** be used for system changes, not reading data.. [...]

Due to the scanning permission involved, the command sudo will be used. However, it is not a complete command procedure shown in the figure. Here, the required command is simply passed to the Terminal unit for execution, and the terminal unit will automatically solve the process of entering the password.

bigtreetech, 2020-11-11:
> @bigtreetech @Shine6Z If I want to recompile my changes for testing, what do I need to run from the rPi ? The below lines ?
> 
> `cd ~/OctoBTT qmake OctoBTT.pro if [ ! -f "$OctoBTT" ]; then rm OctoBTT make clean fi make -j$(cat /proc/cpuinfo | grep processor | wc -l)`

Right, you can execute this command to recompile the boot:
> `cd ~/OctoBTT && qmake OctoBTT.pro && rm OctoBTT && make clean && make -j$(cat /proc/cpuinfo | grep processor | wc -l) && sudo reboot`

bigtreetech, 2020-11-11:
> I have forked the build, and will make some security changes, and do a PR. I'm not that into coding, so I won't be much help with anything outside of security stuff, and recommendations..

Involving safety I can adjust to a more reasonable way to solve this kind of problem, the current operation permissions not immediately remove permissions required demand, this also involves dynamic loading of USB storage devices, but does not rule out future will take a more reasonable way to deal with these problems, thank you for your feedback, also hope that you will continue to focus on, do better suggestion for our improvement.

bigtreetech, 2020-11-11:
> Here is the video for the wifi issues. It looks like it's just hanging, and doesn't do anything at all. I am running on a rPi 4 4GB.
> 
> https://drive.google.com/file/d/1QVfl7Mkzt5qdeotWIxBuVq4aAmaWnNR2/view?usp=sharing

Combined with the overall content, video interface freeze reason is caused by the console password is not correct, my side to freeze the aim is to avoid multiple console driver execution at the same time lead to chaos, so I'll be frozen when entering the video interface interface scanning devices are available, and after the scan will remove freeze, I'm sorry to bring you problems, I will prioritize the unfriendly interface freeze.

Will-wastelander, 2020-11-11:
I was able to resolve the wifi issue w/ my OctoBTT file in sudoers.d. This gives OctoBTT access to run iwlist, iwconfig, wpa_cli, and ifconfig. If there are any other sudo commands that I have missed, please let me know, and I will get them added. Then we no longer need to provide a password for those specific commands to be ran.

Will-wastelander, 2020-11-11:
Would you mind explaining what is going on with the USB mounting stuff ? Why is it needed, what is the function/purpose of it ? I think there is a way to auto-mount USB drives, as it happens when you insert a USB drive when running Xorg.

Will-wastelander, 2020-11-11:
Here are some articles about automounting.

[Automount USB](https://help.ubuntu.com/community/Mount/USB)
[Automount on Ubuntu](https://www.linuxuprising.com/2019/04/automatically-mount-usb-drives-on.html)
[Mount using Autofs](https://linuxconfig.org/automatically-mount-usb-external-drive-with-autofs)

bigtreetech, 2020-11-12:
I would like to have full bash console capability in OctoBTT so that DIY users do not lose their console access because they are using OctoBTT. There are still some issues with introducing Bash, so I'm still looking for a more sensible solution; As for the USB storage device mount problem, I tried the automatic mount scheme before, but for some reason, the mount function did not seem to work, so I took it on OctoBTT and I used the Bash console command to scan and mount the USB storage device.

Will-wastelander, 2020-11-12:
Full bash console is not something a dashboard screen should have. That's what SSH is for. There is no reason a dashboard should ever have full control of a system. 

Ok, but what is the USB storage used for ? autofs, should be able to mount the USB devices, in theory.

I can install and test this if you would like.

Will-wastelander, 2020-11-12:
apt info autofs
Package: autofs
Version: 5.1.2-4
Priority: optional
Section: utils
Maintainer: Debian QA Group <packages@qa.debian.org>
Installed-Size: 2,571 kB
Provides: autofs5
Depends: libc6 (>= 2.17), libxml2 (>= 2.7.4), ucf
Recommends: nfs-common, kmod | module-init-tools, e2fsprogs
Breaks: autofs5 (<< 5.0.6-1~)
Replaces: autofs5 (<< 5.0.6-1~)
Homepage: http://www.kernel.org/pub/linux/daemons/autofs/v5/
Download-Size: 406 kB
APT-Sources: http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages
Description: kernel-based automounter for Linux
 Autofs controls the operation of the automount daemons. The
 automount daemons automatically mount filesystems when they
 are used and unmount them after a period of inactivity. This
 is done based on a set of pre-configured maps.
 .
 The kernel automounter implements an almost complete SunOS
 style automounter under Linux. A recent version of the kernel
 autofs4 module (builtin or separate) is required.
 .
 This is the autofs daemon.

Will-wastelander, 2020-11-12:
You have to remember that security needs to be #1 priority.. Unless BTT wants to be liable for a customer being hacked.

bigtreetech, 2020-11-12:
The dashboard has console access only so that the Raspberry PI can be completely independent of the remote control mode of the computer to achieve full capability, of course, for the raspberry PI only printer system this feature is really redundant, I can completely consider cancelling this feature.
For the use of USB storage devices main consideration is to be able to directly via USB connect directly to raspberries pie print file import and management, without the need to rely on external computer to operate, in the design of future include model can be acquired through USB storage devices to directly implement raspberries pie ability and more complete section can make the printer operation more concise, meaningful function, of course, if you like to help me to install the test we'd like that very much, also thank you very much for your support.

Shine6Z, 2020-11-12:
Dear Will-wastelander:
Hello, we sincerely invite you to become a member of our internal testing team, during which you will get our preferential testing privileges for all kinds of new products for free. If you are interested, you can send your contact information to our email (contact@biqu3d.com), and we will have staff to contact and interact with you. Wish us a pleasant cooperation.

Will-wastelander, 2020-11-12:
@Shine6Z Email has been sent. And unless you want a bunch of other people potentially emailing saying they are me. You should remove contact, or delete post. ;)

Shine6Z, 2020-11-12:
> @Shine6Z Email has been sent. And unless you want a bunch of other people potentially emailing saying they are me. You should remove contact, or delete post. ;)

Ok, we have received your email. We will communicate with you in the email about the internal test of new product deployment in the future.