bigtreetech/OctoBTT #25: No connection to skr 1.4 turbo

CRYSTALK3R, 2020-11-25:
My raspberry doesn't want to connect to my skr 1.4 turbo.  Which settings do I have to make in marlin if I want to establish a connection?  Latest marlin version is installed (nightly build)

bigtreetech, 2021-01-13:
We have tested that the motherboard of SKR V1.4 Turbo can be directly connected to work, we recommend you to try again, if there is still a problem or the purchase of other products from our company has problems, you can contact our after-sales service to provide technical support for you by email: support01@bigtree-tech.com

Arnie3d, 2022-02-06:
Same here! neither detected on the PC nor can it read sd card tried new card was booting for pc then stopped booting also SD on board nothing

Arnie3d, 2022-02-06:
what now.... also spended alot of time and money SKR 1.4 Turbo doesnt want to boot anymore that sucks any bugs we dont know about?