bigtreetech/OctoBTT #28: OctoBTT loads to black screen

dkalinai, 2020-12-30:
My issue is as per the image attached. Prior tohis i see white screen, then the color pallete, then the console print out s, and then boots to this with a small cursor on top left it seems. I followed instructions, saw no issues where can i look to troubleshoot the issue? Bjust got my v 1 BTT PITFT50

bigtreetech, 2021-01-13:
After testing, we confirm that the installation and operation of OctoBTT are all normal. You can try again. If there are still problems, we suggest you provide us with the operation video for observation.

dkalinai, 2021-01-13:
So i did reinstall Octopi and Octoprint andn ow it works indeed, but seems like it was impossible for me to get working on an already running octopi 32bit on raspberry pii4, on 64bit somehow started working. feel free to close, but be aware that there seems to be some issue, i followed the steps 4 times and 2 time it worked on a new sd card and 2 times failed with running octoprint. go figure.

closed 2021-01-13 by dkalinai

bigtreetech, 2021-01-13:
At present, we do not run on the 64-bit operating system, and there may be a big difference in the components on the 64-bit system, so it is not recommended to run directly on the 64-bit operating system, which may bring many unknown exceptions to you. Of course, if you want to run on a 64-bit OS, you can try it out and ask questions, and we welcome your questions.