bigtreetech/OctoBTT #29: High cpu usage

gfenglo, 2021-01-01:
octobtt high cpu usage in idle. around 40% in idle

bigtreetech, 2021-01-13:
The CPU performance loss is normal, which is required by OctoBTT for continuous interface interaction to obtain accurate operating status, and the subsequent performance optimization will continue.

gfenglo, 2021-01-13:
Compared to touch UI, Octodash, and Octoscreen. They use less than 10% CPI usage in idle.

bigtreetech, 2021-01-13:
By contrast, our UI improves the user experience by sacrificing performance. Of course, such a large performance gap is related to the software architecture design we take, so we will improve the actual operation efficiency in the subsequent upgrades, so as to optimize the transition loss on performance.

gfenglo, 2021-01-15:
Are you going to send me another board, or do I have to buy another board?

gfenglo, 2021-01-15:
Sorry, wrong reply.

bigtreetech, 2021-05-05:
Hello, I have solved the problem of excessive CPU occupation. At present, as long as the UI does not stay on the camera monitoring page, there will not be a large load on CPU. Compared with the previous version, the current CPU optimization only takes up 10% of the previous one. If no new questions come up within a week, I will close the current topic.

gfenglo, 2021-05-05:
Sorry, I give up on OctoBTT.

closed 2021-05-08 by bigtreetech