bigtreetech/OctoBTT #32: OctoBTT connection to BiQu BX printer

wolfgangdecker, 2021-04-09:
I received one of the production models and attached a Raspberry Pi Model 3 B+ to the main board. I only seem to get connection to the printer functions if I run a USB cable from the Pi to the front USB port in the printer. In the previews published on YouTube (e.g. by Teaching Tech) it seems that the Pi would connect directly with the printer through the internal serial port. 

What is the problem with my model?

bigtreetech, 2021-04-13:
Thank you for your careful discovery. We cannot completely guarantee that there are no mistakes, but we will learn lessons and correct the known mistakes in time.

bigtreetech, 2021-04-13:
The video shows the earlier version of the development, in the current version of the development can be implemented through the Octo BTT screen to Octoprint to achieve the basic function of the operation; On the BX machine, our machine is not connected to Raspberry Pi through serial port, but directly connected to Raspberry Pi through microUSB or Type C port of the host machine, while Octo BTT will automatically realize the automatic docking between Octoprint and the machine.

wolfgangdecker, 2021-04-13:
Thank you for getting back, appreciate the answer. It is a little ambiguous, though. 

To clarify I have attached two pictures:
[image]
[image]

I had to connect the machine through a USB cable as shown by the yellow circled connections, otherwise OctoBTT does not establish a connection with the printer. You are saying, however, that this should be possible by the connection as shown in the red circle, which is the direct connection with the host machine, which also bring OctoBTT on the screen?

If I only connect as shown in the red circle, OctoBTT is not able to read data from the printer, only if I connect via the connection in the yellow circle. 

Please clarify.

Thank you.

W.

bigtreetech, 2021-04-15:
The connection mode shown in the figure is correct. The red part of the connection cable provides touch input device for the interface of Raspberry Pi, while the yellow part of the connection cable provides connection with the motherboard for Raspberry Pi. Therefore, if Raspberry Pi needs to connect to the motherboard, the yellow part of the connection cable must be installed.

wolfgangdecker, 2021-04-15:
Thank you for the clarification. This was not included in the setup manual, so maybe it should be added.

bigtreetech, 2021-04-23:
Yes, we found that we did have this omission, so we corrected it in the subsequent installation manual of the product. Thank you for your feedback.

closed 2021-04-23 by bigtreetech