bigtreetech/OctoBTT #35: I can't install OctoBTT

cmicheli346, 2021-04-11:
hello, I don't know raspberry pi and following the guide to install OctoBTT I was stuck as shown in the image, could someone explain me how to solve? thank you
[image]

predy3dprint, 2021-04-11:
Firstly try : "sudo apt-get update" (without the quotation mark).

cmicheli346, 2021-04-11:
hi, thanks for the reply.
I did what you suggested and then tried to install again .... same result as before or almost. Only the line changes
[image]

predy3dprint, 2021-04-11:
Hmm... It seems that some repos are not reachable. Especially the QT- Stuff. Is your raspi connected with the internet?
[image]

cmicheli346, 2021-04-11:
yes, first with wifi, now with the ethernet cable

predy3dprint, 2021-04-11:
Sorry. But I found only the following instructions: 

https://awesomeopensource.com/project/bigtreetech/OctoBTT

May be useful for you.

cmicheli346, 2021-04-11:
yes, they are the same ones that I followed

bigtreetech, 2021-04-13:
It looks like the problem is caused by network instability. When you install on a wireless network, you may encounter a failure caused by network instability. At this time, it is recommended that you switch to a wired network for installation.

cmicheli346, 2021-04-13:
ok, the first time i tried to install everything with wifi but after i connected the cable and the problem continues. How can I go about trying to re-download everything? now if I try it tells me that install.sh already exists but if I try to launch it it gives me that error in the end also connected via cable

cmicheli346, 2021-04-18:
no solution ... should I trash everything? it all seems so easy to see the tutorials but I have been stuck for weeks

bigtreetech, 2021-04-20:
It is recommended that you reinstall according to README, because the previous installation failure caused the loss of part of the file may lead to repeated Settings and other problems, so when you run again after the failure of the operation may trigger a second rejection operation, if there is still a failure, you can contact me by email.

E-mail: [developer01@biqu3d.com](developer01@biqu3d.com)

closed 2021-04-23 by bigtreetech