bigtreetech/OctoBTT #36: OctBTT does not install properly

Doodys, 2021-04-20:
My last installation step from install.sh is:

`g++ -Wl,-O1 -o OctoBTT accesscontrol.o cameradialog.o configuration.o controlpanel.o customdialog.o filedialog.o gcodedialog.o inputdialog.o liblog.o main.o mainwindow.o materialctrlpanel.o mylineedit.o octonetwork.o terminaldialog.o wlanconfig.o qrc_assets.o moc_accesscontrol.o moc_cameradialog.o moc_configuration.o moc_controlpanel.o moc_customdialog.o moc_filedialog.o moc_gcodedialog.o moc_inputdialog.o moc_mainwindow.o moc_materialctrlpanel.o moc_terminaldialog.o moc_wlanconfig.o   -lQt5Svg -lQt5Widgets -lQt5Gui -lQt5WebSockets -lQt5Network -lQt5Core -lGLESv2 -lpthread -latomic`

After this step installation stops, RPI reboots and I'm still in console with errors from XORG:

`modprobe: FATAL: Module g2d_23 not found in directory /lib/modules/5.10.17-v7l+
xinit: Connection to X server lost`

I have followed this step by step guide: https://www.youtube.com/watch?v=EQiSkXKjplQ (and some others after encountering above issue).

bigtreetech, 2021-04-22:
The installation method in the video is the old version, the new version has been simplified, you can install according to the README, if you have any questions, you can contact me via email: developer01@biqu3d.com

Doodys, 2021-04-22:
Yeah, README was my first shot. After getting above result I've retried with clean OS and used this older tutorial without any success. If there's any known issue why this stays in console mode instead of jumping into graphical UI - talking about it in here will be better - maybe you won't get the same ticket again 👍

bigtreetech, 2021-04-23:
OctoBTT is a dedicated graphics operating system for Octoprint on RaspberryPi. Booting from the console is unavoidable during installation, and there may still be a lot of problems during use. Above all, we are constantly improving, graphical operating mode as you say, we are also thinking about graphical guide the installation, it can reduce the difficulty of the users in the installation process and get more operating functions, of course this also need time to research and development, so has been published in the current installation tutorial compared to the earlier version is more simple and easy to perform; At present, our improvement is committed to the upgrade of the use experience. In the aspect of bootstrap installation, we have adopted a more concise command to achieve bootstrap installation. So we won't be able to do a graphical boot system on our own in the near term, but it's definitely in our plans.

bigrizzo, 2021-07-15:
> Yeah, README was my first shot. After getting above result I've retried with clean OS and used this older tutorial without any success. If there's any known issue why this stays in console mode [...]

I get the same FATAL error you do. I can't get it to start the GUI. Did you manage to figure out how to get it working?