bigtreetech/OctoBTT #37: Crashing to console

rlmarin, 2021-04-24:
Did a 6 hour print over night no issue.. Started a 10 hour print.. 4 hours in and octobtt crashed to console.. printer is still running as octoprint is still running but on the screen what I see is the octoprint console..

bigtreetech, 2021-05-04:
May I ask which interface did you stop at in the final printing process and then crash?

wkoetz, 2021-05-04:
Hello, I used the BTT graphical interface with babystepping.Z-Offset was default (0). I started a print and adjusted the Z-Offset with babystepping. Stored it to EEPROM. Then I canceled this print and started it again, in mind that Z-Offset should be OK now. Then the nozzle crashes into the bed...Kind regardsWilfried Kötz, 3d-abeitsplatz.deBIQU Reseller in Germany, seehttps://www.3d-arbeitsplatz.de/BIQU-BX-Direct-Extruder-FDM-3D-Drucker/BIQUBXVon meinem/meiner Galaxy gesendet

rlmarin, 2021-05-04:
I left it on to view the Camera as my printer is in a closed cabinet

bigtreetech, 2021-05-05:
> I left it on to view the Camera as my printer is in a closed cabinet
OK, I have determined the cause of the problem, and I will issue an update patch for you to update OCTOBTT to solve the history retention problem.

bigtreetech, 2021-05-05:
Hello, among the products sold by our company, only the B1 and BX models have the ability to save babystep data, but the precondition is that the machine firmware must be our official firmware, because the Marin firmware does not have the ability to save babystep data.

bigtreetech, 2021-05-06:
> I left it on to view the Camera as my printer is in a closed cabinet

Hello, you can update your OCTOBTT by following the command below; The usage is similar to the previous installation, which you can refer to the README for details:
wget --no-check-certificate https://raw.githubusercontent.com/bigtreetech/OctoBTT/master/upgrade.sh && sudo /bin/sh upgrade.sh