bigtreetech/OctoBTT #40: OctBtt blocks CPU

zetti12345, 2021-05-13:
Hi,
OctoBtt runs permanently at 100% Cpu level and blocks everything. Also crashes the print because of blocking.

[image]

bigtreetech, 2021-05-17:
Can you tell me where you left the UI at that time, resulting in excessive CPU usage, so that I can determine the location of the exception code.

Will-wastelander, 2021-05-22:
I can confirm that this is happening with my test system also. Screen is sitting at main screen.

[image]

Will-wastelander, 2021-05-22:
[image]

gawainXX, 2021-06-16:
I'm encountering this as well, OctoBTT is consuming 100% of a single CPU Cure (tends to jump between core 1 through 3) which is causing overall issues with the WEBUI and print reliability.

gawainXX, 2021-06-19:
Tried updating. Reinstalling octoprint and Octobtt to nonavail. Switched to octodash which I'm liking a lot more.