bigtreetech/OctoBTT #42: Is --no-check-certificate necessary?

dxgldotorg, 2021-05-23:
The assets are downloaded from raw.githubusercontent.com which has valid SSL certificates; if a certificate error occurs that means there is something wrong.

Or is the "Great Firewall" damaging your connections?

bigtreetech, 2021-06-07:
However, it worked fine in our installation tests, unless it was caused by a network failure, if you failed to run, you can try to cancel the run parameter.

dxgldotorg, 2021-06-07:
Adding --no-check-certificate bypasses the security features of HTTPS, and if it isn't required should be omitted.

Shine6Z, 2021-07-19:
Of course, in order to ensure that everyone can use it normally, I don't think this option should be cancelled, because there is no reason for this option to harm the interests of users. If there is, I will change other methods to ensure the safety of users. If you have a better way, I will introduce it into the project after testing confirmation.