bigtreetech/OctoBTT #45: Install scripts are not working...

ljohnsonnsb, 2021-06-28:
I am trying to install OctoBTT  on your  5" TFT for Rasp... and it fails completely. Here are the steps I followed from the 'Readme' on this page. 
Download and install Octoprint on a 32gb class 10 sd.
Access the Pi via ssh
Executed:  'sudo apt upgrade'  then 'sudo apt update'  both executed without errors.
Reboot the Pi 
Access and setup Octoprint ,  performed required Octoprint updates. 
Reboot the Pi,  It now shows Octoprint Version 1.6.1 and Octopi Version 0.18.0 
Access the Pi via ssh and run your command to install 'rm -f startup.sh ... "
The script reboots at the end and I get this
[image]
I examined the /etc/rc.local file and found this reference to OctoBTT  'su -l pi -c "cd /home/pi/OctoBTT && ./OctoBTT -platform eglfs" ' 
I ran it with sudo and got this response "-bash: ./OctoBTT: No such file or directory " 
That is where I am now, it appears the is no executable for OctoBTT. 
Where do I go from here.

Temneanu80, 2021-07-18:
Hi, did you managed to get install OctoBTT, i have the same problem and i didn't find any fix anywhere?
Thank you

Shine6Z, 2021-07-19:
run cmdline : ls /home/pi/OctoBTT
I need to confirm that the installation of the program is complete.

Shine6Z, 2021-07-19:
This version requires OctoPi at 0.17.x (requires manual page initialization and any user login when initializing the user).

Will-wastelander, 2021-08-29:
OctoBTT is not limited by the OS version, but by OctoPrint version, due to security that has been put in place by OctoPrint. I have OctoBTT running on OctoPi 0.18.0 with OctoPrint 1.6.1.
```
login as: pi
pi@192.168.4.30's password:
Linux octoBTT-test 5.10.17-v7l+ #1414 SMP Fri Apr 30 13:20:47 BST 2021 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Sun Aug 22 17:36:07 2021

------------------------------------------------------------------------------
Access OctoPrint from a web browser on your network by navigating to any of:

    http://octoBTT-test.local
    http://192.168.4.30

https is also available, with a self-signed certificate.
------------------------------------------------------------------------------
This image comes without a desktop environment installed because it's not
required for running OctoPrint. If you want a desktop environment you can
install it via

    sudo /home/pi/scripts/install-desktop
------------------------------------------------------------------------------
OctoPrint version : 1.6.1
OctoPi version    : 0.18.0
------------------------------------------------------------------------------
```

bigtreetech, 2021-08-30:
On a beta release, we have solved the permission control problem:
https://github.com/bigtreetech/OctoBTT_V2_GUI_Sigma
But the front end still lacks file management, so we're still figuring it out.