bigtreetech/OctoBTT #46: Missing icon files

SomeRandomDave, 2021-07-25:
I tried to install, it fails with complaints about missing files for assets/icon/BingoTouch_Logo_B.svg and assets/icon/BingoTouch_Logo_H.svg, which I cannot find in the repo (even in older versions). Was this perhaps something a developer was tinkering with and forgot to check in?

--
Download OctoBTT ...
Cloning into 'OctoBTT'...
remote: Enumerating objects: 724, done.
remote: Counting objects: 100% (348/348), done.
remote: Compressing objects: 100% (263/263), done.
remote: Total 724 (delta 162), reused 222 (delta 82), pack-reused 376
Receiving objects: 100% (724/724), 10.97 MiB | 2.28 MiB/s, done.
Resolving deltas: 100% (299/299), done.
Done.
Installing OctoBTT ...
Info: creating stash file /home/pi/OctoBTT/.qmake.stash
RCC: Error in 'assets.qrc': Cannot find file 'assets/icon/BingoTouch_Logo_B.svg'
RCC: Error in 'assets.qrc': Cannot find file 'assets/icon/BingoTouch_Logo_H.svg'
rm: cannot remove 'OctoBTT': No such file or directory
rm -f qrc_assets.cpp
rm -f moc_predefs.h
rm -f moc_accesscontrol.cpp moc_cameradialog.cpp moc_configuration.cpp moc_controlpanel.cpp moc_customdialog.cpp moc_filedialog.cpp moc_gcodedialog.cpp moc_inputdialog.cpp moc_mainwindow.cpp moc_materialctrlpanel.cpp moc_terminaldialog.cpp moc_wlanconfig.cpp
rm -f ui_cameradialog.h ui_configuration.h ui_controlpanel.h ui_filedialog.h ui_gcodedialog.h ui_inputdialog.h ui_mainwindow.h ui_materialctrlpanel.h ui_terminaldialog.h ui_wlanconfig.h
rm -f accesscontrol.o cameradialog.o configuration.o controlpanel.o customdialog.o filedialog.o gcodedialog.o inputdialog.o liblog.o main.o mainwindow.o materialctrlpanel.o mylineedit.o octonetwork.o terminaldialog.o wlanconfig.o qrc_assets.o moc_accesscontrol.o moc_cameradialog.o moc_configuration.o moc_controlpanel.o moc_customdialog.o moc_filedialog.o moc_gcodedialog.o moc_inputdialog.o moc_mainwindow.o moc_materialctrlpanel.o moc_terminaldialog.o moc_wlanconfig.o
rm -f *~ core *.core
/usr/lib/qt5/bin/uic cameradialog.ui -o ui_cameradialog.h
/usr/lib/qt5/bin/uic configuration.ui -o ui_configuration.h
/usr/lib/qt5/bin/uic controlpanel.ui -o ui_controlpanel.h
/usr/lib/qt5/bin/uic filedialog.ui -o ui_filedialog.h
/usr/lib/qt5/bin/uic gcodedialog.ui -o ui_gcodedialog.h
/usr/lib/qt5/bin/uic inputdialog.ui -o ui_inputdialog.h
/usr/lib/qt5/bin/uic mainwindow.ui -o ui_mainwindow.h
/usr/lib/qt5/bin/uic materialctrlpanel.ui -o ui_materialctrlpanel.h
/usr/lib/qt5/bin/uic terminaldialog.ui -o ui_terminaldialog.h
/usr/lib/qt5/bin/uic wlanconfig.ui -o ui_wlanconfig.h
g++ -c -pipe -O2 -std=gnu++11 -Wall -W -D_REENTRANT -fPIC -DQT_DEPRECATED_WARNINGS -DQT_NO_DEBUG -DQT_SVG_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -isystem /usr/include/arm-linux-gnueabihf/qt5 -isystem /usr/include/arm-linux-gnueabihf/qt5/QtSvg -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWidgets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtGui -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWebSockets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtNetwork -isystem /usr/include/arm-linux-gnueabihf/qt5/QtCore -I. -I. -I/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -o accesscontrol.o accesscontrol.cpp
g++ -c -pipe -O2 -std=gnu++11 -Wall -W -D_REENTRANT -fPIC -DQT_DEPRECATED_WARNINGS -DQT_NO_DEBUG -DQT_SVG_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -isystem /usr/include/arm-linux-gnueabihf/qt5 -isystem /usr/include/arm-linux-gnueabihf/qt5/QtSvg -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWidgets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtGui -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWebSockets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtNetwork -isystem /usr/include/arm-linux-gnueabihf/qt5/QtCore -I. -I. -I/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -o cameradialog.o cameradialog.cpp
g++ -c -pipe -O2 -std=gnu++11 -Wall -W -D_REENTRANT -fPIC -DQT_DEPRECATED_WARNINGS -DQT_NO_DEBUG -DQT_SVG_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -isystem /usr/include/arm-linux-gnueabihf/qt5 -isystem /usr/include/arm-linux-gnueabihf/qt5/QtSvg -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWidgets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtGui -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWebSockets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtNetwork -isystem /usr/include/arm-linux-gnueabihf/qt5/QtCore -I. -I. -I/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -o configuration.o configuration.cpp
g++ -c -pipe -O2 -std=gnu++11 -Wall -W -D_REENTRANT -fPIC -DQT_DEPRECATED_WARNINGS -DQT_NO_DEBUG -DQT_SVG_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -isystem /usr/include/arm-linux-gnueabihf/qt5 -isystem /usr/include/arm-linux-gnueabihf/qt5/QtSvg -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWidgets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtGui -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWebSockets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtNetwork -isystem /usr/include/arm-linux-gnueabihf/qt5/QtCore -I. -I. -I/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -o controlpanel.o controlpanel.cpp
g++ -c -pipe -O2 -std=gnu++11 -Wall -W -D_REENTRANT -fPIC -DQT_DEPRECATED_WARNINGS -DQT_NO_DEBUG -DQT_SVG_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -isystem /usr/include/arm-linux-gnueabihf/qt5 -isystem /usr/include/arm-linux-gnueabihf/qt5/QtSvg -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWidgets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtGui -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWebSockets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtNetwork -isystem /usr/include/arm-linux-gnueabihf/qt5/QtCore -I. -I. -I/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -o customdialog.o customdialog.cpp
g++ -c -pipe -O2 -std=gnu++11 -Wall -W -D_REENTRANT -fPIC -DQT_DEPRECATED_WARNINGS -DQT_NO_DEBUG -DQT_SVG_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -isystem /usr/include/arm-linux-gnueabihf/qt5 -isystem /usr/include/arm-linux-gnueabihf/qt5/QtSvg -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWidgets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtGui -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWebSockets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtNetwork -isystem /usr/include/arm-linux-gnueabihf/qt5/QtCore -I. -I. -I/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -o filedialog.o filedialog.cpp
g++ -c -pipe -O2 -std=gnu++11 -Wall -W -D_REENTRANT -fPIC -DQT_DEPRECATED_WARNINGS -DQT_NO_DEBUG -DQT_SVG_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -isystem /usr/include/arm-linux-gnueabihf/qt5 -isystem /usr/include/arm-linux-gnueabihf/qt5/QtSvg -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWidgets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtGui -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWebSockets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtNetwork -isystem /usr/include/arm-linux-gnueabihf/qt5/QtCore -I. -I. -I/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -o gcodedialog.o gcodedialog.cpp
g++ -c -pipe -O2 -std=gnu++11 -Wall -W -D_REENTRANT -fPIC -DQT_DEPRECATED_WARNINGS -DQT_NO_DEBUG -DQT_SVG_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -isystem /usr/include/arm-linux-gnueabihf/qt5 -isystem /usr/include/arm-linux-gnueabihf/qt5/QtSvg -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWidgets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtGui -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWebSockets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtNetwork -isystem /usr/include/arm-linux-gnueabihf/qt5/QtCore -I. -I. -I/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -o inputdialog.o inputdialog.cpp
g++ -c -pipe -O2 -std=gnu++11 -Wall -W -D_REENTRANT -fPIC -DQT_DEPRECATED_WARNINGS -DQT_NO_DEBUG -DQT_SVG_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -isystem /usr/include/arm-linux-gnueabihf/qt5 -isystem /usr/include/arm-linux-gnueabihf/qt5/QtSvg -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWidgets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtGui -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWebSockets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtNetwork -isystem /usr/include/arm-linux-gnueabihf/qt5/QtCore -I. -I. -I/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -o liblog.o liblog.cpp
[... cut ...] QtGui -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWebSockets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtNetwork -isystem /usr/include/arm-linux-gnueabihf/qt5/QtCore -I. -I. -I/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -o wlanconfig.o wlanconfig.cpp
/usr/lib/qt5/bin/rcc -name assets assets.qrc -o qrc_assets.cpp
RCC: Error in 'assets.qrc': Cannot find file 'assets/icon/BingoTouch_Logo_B.svg'
make: *** [Makefile:582: qrc_assets.cpp] Error 1
make: *** Waiting for unfinished jobs....

SomeRandomDave, 2021-07-31:
Had a chance to revisit this. Files were removed by user bigtreetech on 23 June (commit 1cc03232d0864c712c91e931434ad6dc0a00d97d) with a comment about removing a test file. May want to either update the build files to not call them or undelete them.

bigtreetech, 2021-08-17:
Fix:
https://github.com/bigtreetech/OctoBTT/commit/d690a82b11a93fc60c0a64d7d8d50bac0732d2bc

closed 2021-08-17 by bigtreetech